patrick3399/Hi-Link_mmWave_Radar_ESPHome #1: esp8266 Can't I use it?

qkddn49, 2022-07-26:
It stops when the motion is detected
The presence is normal

patrick3399, 2022-07-26:
Try to delete all sensor, only left UART_TEXT, then check the text sensor work status.

PhillipRt, 2022-07-27:
I could not get it to work on an D1-mini, I think either UART is to slow or the whole chip has to little processing power. Same configuration works flawless with an esp32

qkddn49, 2022-07-27:
Thank you for your answer

patrick3399, 2022-07-27:
Try to use Hardware UART or Software UART on 8266, UART is the most important part in this program.

patrick3399, 2022-07-27:
Use this config to test ESP8266 UART
https://github.com/patrick3399/Hi-Link_mmWave_Radar_ESPHome/blob/main/ESP8266_Standard_UART.yaml

qkddn49, 2022-07-27:
The signal is normal
--------------------------------------------------------------------------

INFO Reading configuration /config/esphome/ld1115h-esp8266.yaml...
INFO Starting log output from esp-radar.local using esphome API
INFO Successfully connected to esp-radar.local
[22:02:37][I][app:102]: ESPHome version 2022.6.2 compiled on Jul 27 2022, 21:56:51
[22:02:37][C][wifi:491]: WiFi:
[22:02:37][C][wifi:353]:   Local MAC: xx:xx:xx:xx:xx:xx
[22:02:37][C][wifi:354]:   SSID: 'xxxxxxxx'[redacted]
[22:02:37][C][wifi:355]:   IP Address: 192.168.0.154
[22:02:37][C][wifi:356]:   BSSID: xx:xx:xx:xx:xx:xx[redacted]
[22:02:37][C][wifi:358]:   Hostname: 'esp-radar'
[22:02:37][C][wifi:360]:   Signal strength: -47 dB ▂▄▆█
[22:02:37][C][wifi:364]:   Channel: 11
[22:02:37][C][wifi:365]:   Subnet: 255.255.255.0
[22:02:37][C][wifi:366]:   Gateway: 192.168.0.1
[22:02:37][C][wifi:367]:   DNS1: 168.126.63.1
[22:02:37][C][wifi:368]:   DNS2: 168.126.63.2
[22:02:37][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 11416'
[22:02:37][C][logger:275]: Logger:
[22:02:37][C][logger:276]:   Level: DEBUG
[22:02:37][C][logger:277]:   Log Baud Rate: 0
[22:02:37][C][logger:278]:   Hardware UART: UART0
[22:02:37][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 14996'
[22:02:37][C][uart.arduino_esp8266:102]: UART Bus:
[22:02:37][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[22:02:37][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[22:02:37][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[22:02:37][C][uart.arduino_esp8266:108]:   Baud Rate: 115200 baud
[22:02:37][C][uart.arduino_esp8266:109]:   Data Bits: 8
[22:02:37][C][uart.arduino_esp8266:110]:   Parity: NONE
[22:02:37][C][uart.arduino_esp8266:111]:   Stop bits: 1
[22:02:37][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[22:02:37][C][mdns:084]: mDNS:
[22:02:37][C][mdns:085]:   Hostname: esp-radar
[22:02:37][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 13601'
[22:02:37][C][ota:085]: Over-The-Air Updates:
[22:02:37][C][ota:086]:   Address: esp-radar.local:8266
[22:02:37][C][api:138]: API Server:
[22:02:37][C][api:139]:   Address: esp-radar.local:6053
[22:02:37][C][api:143]:   Using noise encryption: NO
[22:02:37][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 10737'
[22:02:37][C][serial.text_sensor:029]: Serial Sensor 'ESP Radar UART Text'
[22:02:37][C][serial.text_sensor:029]:   Icon: 'mdi:format-text'
[22:02:37][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 6316'
[22:02:37][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 4326'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 4118'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 4118'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 4006'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 2556'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 1046'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 838'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 880'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 838'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 682'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 501'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 360'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'mov, 2 199'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 2 2362'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 2 2370'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 2 2370'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 2 2370'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 2 2370'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 2 2357'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 2 2337'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 2 2337'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 2 2337'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 2 2290'
[22:02:38][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 2 2244'
[... 508 lines cut ...]
[22:02:58][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 6 6863'
[22:02:58][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 6 6863'
[22:02:58][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 6 6863'
[22:02:58][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 6 6442'
[22:02:58][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 6 6442'
[22:02:58][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 7 6442'
[22:02:58][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 8 5953'
[22:02:58][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 8 5721'
[22:02:58][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 9 5721'
[22:02:58][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 9 5721'
[22:02:58][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 9 5721'
[... cut ...]
[22:02:58][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 5 5867'
[22:02:58][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 4 5639'
[22:02:59][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 4 5639'
[22:02:59][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 4 5639'
[22:02:59][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 4 5639'
[22:02:59][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 4 5639'
[22:02:59][D][text_sensor:067]: 'ESP Radar UART Text': Sending state 'occ, 5 5639'

patrick3399, 2022-07-27:
Try this again.
https://github.com/patrick3399/Hi-Link_mmWave_Radar_ESPHome/blob/main/LD1115H/ESP8266-LD1115H.yaml

qkddn49, 2022-07-27:
INFO Reading configuration /config/esphome/ld115h-esp8266.yaml...
Failed config

interval: [source /config/esphome/ld115h-esp8266.yaml:51]
  - interval: 1s
    setup_priority: -200.0
    then: 
      - 
        Couldn't find ID 'LD1115H_Clear_Time'. Please check you have defined an ID with that name in your configuration. These IDs look similar: "LD1115H_Last_Time", "LD1115H_Last_Mov_Time", "LD1115H_Clearence_Status".
        lambda: !lambda |-
          if ((time(NULL)-id(LD1115H_Last_Time))>id(LD1115H_Clear_Time).state) {
            if ((id(LD1115H_Clearence_Status) == false) || (id(LD1115H_Occupancy).state != "Clearance")) {
              id(LD1115H_Occupancy).publish_state("Clearance");
              id(LD1115H_Clearence_Status) = true;
            }
text_sensor.serial: [source /config/esphome/ld115h-esp8266.yaml:75]
  platform: serial
  uart_id: LD1115_UART_BUS
  name: ESP Radar UART Text
  id: LD1115_UART_Text
  icon: mdi:format-text
  internal: False
  on_value:  [source /config/esphome/ld115h-esp8266.yaml:81]
    - [source /config/esphome/ld115h-esp8266.yaml:81]
      then:  [source /config/esphome/ld115h-esp8266.yaml:81]
        - [source /config/esphome/ld115h-esp8266.yaml:81]
          
          Couldn't find ID 'LD1115H_UART_Text'. Please check you have defined an ID with that name in your configuration. These IDs look similar: "LD1115_UART_Text", "LD1115_UART_BUS", "LD1115H_Last_Time".
          lambda: !lambda |-
            if (id(LD1115H_UART_Text).state.substr(0,3) == "occ") {
              id(LD1115H_Signal).publish_state(atof(id(LD1115H_UART_Text).state.substr(7).c_str()));
              id(LD1115H_Direction).publish_state(atof(id(LD1115H_UART_Text).state.substr(5,2).c_str()));
              if ((time(NULL)-id(LD1115H_Last_Mov_Time))>id(LD1115H_Mov_Time).state) {
                id(LD1115H_Occupancy).publish_state("Occupancy");
              }
              id(LD1115H_Last_Time) = time(NULL);
              if (id(LD1115H_Clearence_Status) == true) {
                id(LD1115H_Clearence_Status) = false;
              }
            }
            else if (id(LD1115H_UART_Text).state.substr(0,3) == "mov") {
              id(LD1115H_Signal).publish_state(atof(id(LD1115H_UART_Text).state.substr(7).c_str()));
              id(LD1115H_Direction).publish_state(atof(id(LD1115H_UART_Text).state.substr(5,2).c_str()));
              id(LD1115H_Occupancy).publish_state("Movement");
              id(LD1115H_Last_Mov_Time) = time(NULL);
              id(LD1115H_Last_Time) = time(NULL);
              if (id(LD1115H_Clearence_Status) == true) {
                id(LD1115H_Clearence_Status) = false;
              }
            } [source /config/esphome/ld115h-esp8266.yaml:81]
  disabled_by_default: False
Couldn't find ID 'LD1115H_UART_Text'. Please check you have defined an ID with that name in your configuration. These IDs look similar: "LD1115_UART_Text", "LD1115_UART_BUS", "LD1115H_Last_Time".

Couldn't find ID 'LD1115H_Mov_Time'. Please check you have defined an ID with that name in your configuration. These IDs look similar: "LD1115H_Last_Mov_Time", "LD1115H_Last_Time", "LD1115_UART_Text".

Couldn't find ID 'LD1115H_UART_Text'. Please check you have defined an ID with that name in your configuration. These IDs look similar: "LD1115_UART_Text", "LD1115_UART_BUS", "LD1115H_Last_Time".

Couldn't find ID 'LD1115H_UART_Text'. Please check you have defined an ID with that name in your configuration. These IDs look similar: "LD1115_UART_Text", "LD1115_UART_BUS", "LD1115H_Last_Time".

Couldn't find ID 'LD1115H_UART_Text'. Please check you have defined an ID with that name in your configuration. These IDs look similar: "LD1115_UART_Text", "LD1115_UART_BUS", "LD1115H_Last_Time".

patrick3399, 2022-07-27:
Fix. Try again.

qkddn49, 2022-07-27:
Thank you so much.
It works normally.
-------------------------------------------------------------------------------------------

INFO Reading configuration /config/esphome/ld115h-esp8266.yaml...
INFO Starting log output from esp-radar.local using esphome API
INFO Successfully connected to esp-radar.local
[05:39:14][I][app:102]: ESPHome version 2022.6.2 compiled on Jul 28 2022, 05:33:20
[05:39:14][C][wifi:491]: WiFi:
[05:39:14][C][wifi:353]:   Local 
[05:39:14][C][wifi:354]:   SSID: 
[05:39:14][C][wifi:355]:   IP Address: 192.168.0.154
[05:39:14][C][wifi:356]:   BSSID: 
[05:39:14][C][wifi:358]:   Hostname: 'esp-radar'
[05:39:14][C][wifi:360]:   Signal strength: -51 dB ▂▄▆█
[05:39:14][C][wifi:364]:   Channel: 11
[05:39:14][C][wifi:365]:   Subnet: 255.255.255.0
[05:39:14][C][wifi:366]:   Gateway: 192.168.0.1
[05:39:14][C][wifi:367]:   DNS1: 168.126.63.1
[05:39:14][C][wifi:368]:   DNS2: 168.126.63.2
[05:39:14][D][text_sensor:067]: 'ESP Radar LD1115H UART Text': Sending state 'mov, 3 407'
[05:39:14][D][sensor:124]: 'ESP Radar LD1115H Direction': Sending state 3.00000  with 0 decimals of accuracy
[05:39:14][D][text_sensor:067]: 'ESP Radar LD1115H Occupancy Status': Sending state 'Movement'
[05:39:14][C][logger:275]: Logger:
[05:39:14][C][logger:276]:   Level: DEBUG
[05:39:14][C][logger:277]:   Log Baud Rate: 0
[05:39:14][C][logger:278]:   Hardware UART: UART0
[05:39:14][D][text_sensor:067]: 'ESP Radar LD1115H UART Text': Sending state 'mov, 3 358'
[05:39:14][D][sensor:124]: 'ESP Radar LD1115H Direction': Sending state 3.00000  with 0 decimals of accuracy
[05:39:14][D][text_sensor:067]: 'ESP Radar LD1115H Occupancy Status': Sending state 'Movement'
[05:39:14][C][uart.arduino_esp8266:102]: UART Bus:
[05:39:14][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[05:39:14][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[05:39:14][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[05:39:14][C][uart.arduino_esp8266:108]:   Baud Rate: 115200 baud
[05:39:14][C][uart.arduino_esp8266:109]:   Data Bits: 8
[05:39:14][C][uart.arduino_esp8266:110]:   Parity: NONE
[05:39:14][C][uart.arduino_esp8266:111]:   Stop bits: 1
[05:39:14][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[05:39:14][D][text_sensor:067]: 'ESP Radar LD1115H UART Text': Sending state 'mov, 3 358'
[05:39:14][D][sensor:124]: 'ESP Radar LD1115H Direction': Sending state 3.00000  with 0 decimals of accuracy
[05:39:14][D][text_sensor:067]: 'ESP Radar LD1115H Occupancy Status': Sending state 'Movement'
[05:39:14][C][template.number:050]: Template Number 'ESP Radar LD1115H TH1'
[05:39:14][C][template.number:050]:   Icon: 'mdi:cogs'
[05:39:14][C][template.number:051]:   Optimistic: YES
[05:39:14][C][template.number:052]:   Update Interval: 60.0s
[05:39:14][C][template.number:050]: Template Number 'ESP Radar LD1115H TH2'
[05:39:14][C][template.number:050]:   Icon: 'mdi:cogs'
[05:39:14][C][template.number:051]:   Optimistic: YES
[05:39:14][C][template.number:052]:   Update Interval: 60.0s
[05:39:14][D][text_sensor:067]: 'ESP Radar LD1115H UART Text': Sending state 'occ, 8 2747'
[05:39:14][D][sensor:124]: 'ESP Radar LD1115H Direction': Sending state 8.00000  with 0 decimals of accuracy
[05:39:14][C][template.number:050]: Template Number 'ESP Radar LD1115H Clearence Time'
[05:39:14][C][template.number:050]:   Icon: 'mdi:cogs'
[05:39:14][C][template.number:051]:   Optimistic: YES
[05:39:14][C][template.number:052]:   Update Interval: 60.0s
[05:39:14][D][sensor:124]: 'ESP Radar LD1115H Signal Strength': Sending state 728.87500  with 0 decimals of accuracy
[05:39:14][C][template.number:050]: Template Number 'ESP Radar LD1115H Movement Time'
[05:39:14][C][template.number:050]:   Icon: 'mdi:cogs'
[05:39:14][C][template.number:051]:   Optimistic: YES
[05:39:14][C][template.number:052]:   Update Interval: 60.0s
[05:39:14][C][template.sensor:023]: Template Sensor 'ESP Radar LD1115H Direction'
[05:39:14][C][template.sensor:023]:   State Class: ''
[05:39:14][C][template.sensor:023]:   Unit of Measurement: ''
[05:39:14][C][template.sensor:023]:   Accuracy Decimals: 0
[05:39:14][C][template.sensor:023]:   Icon: 'mdi:radar'
[05:39:14][C][template.sensor:024]:   Update Interval: 60.0s
[05:39:14][D][text_sensor:067]: 'ESP Radar LD1115H UART Text': Sending state 'mov, 2 358'
[05:39:14][D][sensor:124]: 'ESP Radar LD1115H Direction': Sending state 2.00000  with 0 decimals of accuracy
[05:39:14][D][text_sensor:067]: 'ESP Radar LD1115H Occupancy Status': Sending state 'Movement'
[... 95 lines cut ...]
[05:39:15][D][sensor:124]: 'ESP Radar LD1115H Direction': Sending state 2.00000  with 0 decimals of accuracy
[05:39:15][D][text_sensor:067]: 'ESP Radar LD1115H Occupancy Status': Sending state 'Movement'
[05:39:15][D][text_sensor:067]: 'ESP Radar LD1115H UART Text': Sending state 'mov, 2 431'
[05:39:15][D][sensor:124]: 'ESP Radar LD1115H Direction': Sending state 2.00000  with 0 decimals of accuracy
[05:39:15][D][text_sensor:067]: 'ESP Radar LD1115H Occupancy Status': Sending state 'Movement'
[05:39:15][D][sensor:124]: 'ESP Radar LD1115H Signal Strength': Sending state 390.12500  with 0 decimals of accuracy
[05:39:15][D][text_sensor:067]: 'ESP Radar LD1115H UART Text': Sending state 'mov, 3 379'
[05:39:15][D][sensor:124]: 'ESP Radar LD1115H Direction': Sending state 3.00000  with 0 decimals of accuracy
[05:39:15][D][text_sensor:067]: 'ESP Radar LD1115H Occupancy Status': Sending state 'Movement'
[... cut ...]
[05:39:15][D][text_sensor:067]: 'ESP Radar LD1115H Occupancy Status': Sending state 'Movement'
[05:39:15][D][sensor:124]: 'ESP Radar LD1115H Signal Strength': Sending state 391.87500  with 0 decimals of accuracy
[05:39:15][D][text_sensor:067]: 'ESP Radar LD1115H UART Text': Sending state 'mov, 3 363'
[05:39:15][D][sensor:124]: 'ESP Radar LD1115H Direction': Sending state 3.00000  with 0 decimals of accuracy
[05:39:15][D][text_sensor:067]: 'ESP Radar LD1115H Occupancy Status': Sending state 'Movement'

patrick3399, 2022-07-29:
Solve

closed 2022-07-29 by patrick3399

qkddn49, 2022-08-10:
> 해결하다

May I have your email address?
HI-Link is going to request an espome code for HLK-LD2410.
He's asking me.