patrick3399/Hi-Link_mmWave_Radar_ESPHome #4: Explanation of HLK-LD1115H Parameters

crlogic, 2022-08-09:
Hi

Can you please provide the minimum and maximum values for all parameters including a description.

```
16:51:56][D][uart_debug:158]: >>> "get_all\n"
[16:51:56][D][uart_debug:158]: <<< "th1 is 130\r\n"
[16:51:56][D][uart_debug:158]: <<< "th2 is 250\r\n"
[16:51:56][D][uart_debug:158]: <<< "th3 is 1500\r\n"
[16:51:56][D][uart_debug:158]: <<< "ind_min is 2\r\n"
[16:51:56][D][uart_debug:158]: <<< "ind_max is 32\r\n"
[16:51:56][D][uart_debug:158]: <<< "mov_sn is 3\r\n"
[16:51:56][D][uart_debug:158]: <<< "occ_sn is 5\r\n"
[16:51:56][D][uart_debug:158]: <<< "dtime is 5000 mS\r\n"
```
I do not see min/max for any values in the datasheet.  Nor any description for th3, ind_*, etc..

qkddn49, 2022-08-09:
INFO Reading configuration /config/esphome/ld1115h.yaml...
WARNING GPIO2 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from ld1115h.local using esphome API
WARNING Can't connect to ESPHome API for ld1115h.local: Error resolving IP address: [Errno -5] No address associated with hostname
INFO Trying to reconnect to ld1115h.local in the background
INFO Successfully connected to ld1115h.local
[07:18:38][I][app:102]: ESPHome version 2022.6.3 compiled on Aug  9 2022, 13:08:44
[07:18:38][C][status_led:019]: Status LED:
[07:18:38][C][status_led:020]:   Pin: GPIO2
[07:18:38][C][wifi:491]: WiFi:
[07:18:38][C][wifi:353]:   Local MAC: 
[07:18:38][C][wifi:354]:   SSID: 
[07:18:38][C][wifi:355]:   IP Address: 192.168.0.157
[07:18:38][C][wifi:357]:   BSSID: 72:5D:CC:4A:94:AC[redacted]
[07:18:38][C][wifi:358]:   Hostname: 'ld1115h'
[07:18:38][C][wifi:360]:   Signal strength: -77 dB ▂▄▆█
[07:18:38][C][wifi:364]:   Channel: 2
[07:18:38][C][wifi:365]:   Subnet: 255.255.255.0
[07:18:38][C][wifi:366]:   Gateway: 192.168.0.1
[07:18:38][C][wifi:367]:   DNS1: 168.126.63.1
[07:18:38][C][wifi:368]:   DNS2: 168.126.63.2
[07:18:38][C][logger:275]: Logger:
[07:18:38][C][logger:276]:   Level: DEBUG
[07:18:38][C][logger:277]:   Log Baud Rate: 0
[07:18:38][C][logger:278]:   Hardware UART: UART0
[07:18:38][C][i2c.idf:047]: I2C Bus:
[07:18:38][C][i2c.idf:048]:   SDA Pin: GPIO21
[07:18:38][C][i2c.idf:049]:   SCL Pin: GPIO22
[07:18:38][C][i2c.idf:050]:   Frequency: 50000 Hz
[07:18:38][C][i2c.idf:053]:   Recovery: bus successfully recovered
[07:18:38][I][i2c.idf:063]: Results from i2c bus scan:
[07:18:38][I][i2c.idf:069]: Found i2c device at address 0x38
[07:18:38][D][sensor:125]: 'ESP Radar LD1115H Signal Strength': Sending state 162.25000  with 0 decimals of accuracy
[07:18:38][C][template.number:050]: Template Number 'ESP Radar LD1115H TH1'
[07:18:38][C][template.number:050]:   Icon: 'mdi:cogs'
[07:18:38][C][template.number:051]:   Optimistic: YES
[07:18:38][C][template.number:052]:   Update Interval: 60.0s
[07:18:38][C][template.number:050]: Template Number 'ESP Radar LD1115H TH2'
[07:18:38][C][template.number:050]:   Icon: 'mdi:cogs'
[07:18:39][C][template.number:051]:   Optimistic: YES
[07:18:39][C][template.number:052]:   Update Interval: 60.0s
[07:18:39][C][template.number:050]: Template Number 'ESP Radar LD1115H Clearence Time'
[07:18:39][C][template.number:050]:   Icon: 'mdi:cogs'
[07:18:39][C][template.number:051]:   Optimistic: YES
[07:18:39][C][template.number:052]:   Update Interval: 60.0s
[07:18:39][C][template.number:050]: Template Number 'ESP Radar LD1115H Movement Time'
[07:18:39][C][template.number:050]:   Icon: 'mdi:cogs'
[07:18:39][C][template.number:051]:   Optimistic: YES
[07:18:39][C][template.number:052]:   Update Interval: 60.0s
[07:18:39][C][template.sensor:023]: Template Sensor 'ESP Radar LD1115H Spectral line'
[07:18:39][C][template.sensor:023]:   State Class: ''
[07:18:39][C][template.sensor:023]:   Unit of Measurement: ''
[07:18:39][C][template.sensor:023]:   Accuracy Decimals: 0
[07:18:39][C][template.sensor:023]:   Icon: 'mdi:radar'
[07:18:39][C][template.sensor:024]:   Update Interval: 60.0s
[07:18:39][C][template.sensor:023]: Template Sensor 'ESP Radar LD1115H Signal Strength'
[07:18:39][C][template.sensor:023]:   State Class: ''
[07:18:39][C][template.sensor:023]:   Unit of Measurement: ''
[07:18:39][C][template.sensor:023]:   Accuracy Decimals: 0
[07:18:39][C][template.sensor:023]:   Icon: 'mdi:signal-distance-variant'
[07:18:39][C][template.sensor:024]:   Update Interval: 60.0s
[07:18:39][C][template.text_sensor:021]: Template Sensor 'ESP Radar LD1115H Occupancy Status'
[07:18:39][C][template.text_sensor:021]:   Icon: 'mdi:motion-sensor'
[07:18:39][C][template.binary_sensor:018]: Template Binary Sensor 'ESP Radar LD1115H Occupancy or Movement'
[07:18:39][C][template.binary_sensor:018]:   Device Class: 'occupancy'
[07:18:39][C][template.binary_sensor:018]: Template Binary Sensor 'ESP Radar LD1115H Movement'
[07:18:39][C][template.binary_sensor:018]:   Device Class: 'motion'
[07:18:39][C][aht10:135]: AHT10:
[... 612 lines cut ...]
[07:18:50][D][text_sensor:067]: 'ESP Radar LD1115H UART Text': Sending state 'mov, 3 55'
[07:18:50][D][sensor:125]: 'ESP Radar LD1115H Spectral line': Sending state 3.00000  with 0 decimals of accuracy
[07:18:50][D][text_sensor:067]: 'ESP Radar LD1115H Occupancy Status': Sending state 'Movement'
[07:18:50][D][sensor:125]: 'ESP Radar LD1115H Signal Strength': Sending state 57.12500  with 0 decimals of accuracy
[07:18:50][D][text_sensor:067]: 'ESP Radar LD1115H UART Text': Sending state 'mov, 3 47'
[07:18:50][D][sensor:125]: 'ESP Radar LD1115H Spectral line': Sending state 3.00000  with 0 decimals of accuracy
[07:18:50][D][text_sensor:067]: 'ESP Radar LD1115H Occupancy Status': Sending state 'Movement'
[07:18:50][D][text_sensor:067]: 'ESP Radar LD1115H UART Text': Sending state 'mov, 2 47'
[07:18:50][D][sensor:125]: 'ESP Radar LD1115H Spectral line': Sending state 2.00000  with 0 decimals of accuracy
[07:18:50][D][text_sensor:067]: 'ESP Radar LD1115H Occupancy Status': Sending state 'Movement'
[07:18:50][D][text_sensor:067]: 'ESP Radar LD1115H UART Text': Sending state 'mov, 3 54'
[... cut ...]
[07:18:50][D][sensor:125]: 'ESP Radar LD1115H Signal Strength': Sending state 52.62500  with 0 decimals of accuracy
[07:18:50][D][text_sensor:067]: 'ESP Radar LD1115H UART Text': Sending state 'mov, 3 56'
[07:18:50][D][sensor:125]: 'ESP Radar LD1115H Spectral line': Sending state 3.00000  with 0 decimals of accuracy
[07:18:50][D][text_sensor:067]: 'ESP Radar LD1115H Occupancy Status': Sending state 'Movement'
[07:18:50][D][text_sensor:067]: 'ESP Radar LD1115H UART Text': Sending state 'mov, 3 51'
[07:18:50][D][sensor:125]: 'ESP Radar LD1115H Spectral

crlogic, 2022-08-09:
So you don't know?

qkddn49, 2022-08-09:
p{margin-top:0;margin-bottom:0}
    

    Message ID: ***@***.***>

    

Isn't LD1115h originally without th3 value?

Doesn't show up like example data

crlogic, 2022-08-09:
> Isn't LD1115h originally without th3 value?

Mine has it. Do you know what it represents?

qkddn49, 2022-08-09:
p{margin-top:0;margin-bottom:0}
    

    I don't understand.Only th1 and th2 are marked on the sales page, so I checked only this parthttps://ko.aliexpress.com/item/1005004139492358.html?spm=a2g0s.8937460.0.0.29132e0eoxHA2e&gatewayAdapt=glo2korth1 = motion detection sensitivityth2 = presence detection sensitivity

crlogic, 2022-08-09:
> Only th1 and th2 are marked on the sales page

Yes that is why I am asking..

Send `get_all` to the HLK-LD1115H and it returns, th1, th2, th3, ind_min, ind_max...
```
16:51:56][D][uart_debug:158]: >>> "get_all\n"
[16:51:56][D][uart_debug:158]: <<< "th1 is 130\r\n"
[16:51:56][D][uart_debug:158]: <<< "th2 is 250\r\n"
[16:51:56][D][uart_debug:158]: <<< "th3 is 1500\r\n"
[16:51:56][D][uart_debug:158]: <<< "ind_min is 2\r\n"
[16:51:56][D][uart_debug:158]: <<< "ind_max is 32\r\n"
[16:51:56][D][uart_debug:158]: <<< "mov_sn is 3\r\n"
[16:51:56][D][uart_debug:158]: <<< "occ_sn is 5\r\n"
[16:51:56][D][uart_debug:158]: <<< "dtime is 5000 mS\r\n"
```
I can set th3 by sending `th3=xxx` and it saves the value.  So what does it do??

qkddn49, 2022-08-10:
p{margin-top:0;margin-bottom:0}
    

    I asked the dealer

"th3 is for engineering testing."

I received an answer that

crlogic, 2022-08-10:
Can you ask how it works?

qkddn49, 2022-08-10:
p{margin-top:0;margin-bottom:0}
    

    ***@***.*** is the contact email address.You can talk to them.

crlogic, 2022-08-10:
> ***@***.*** is the contact email address.You can talk to them

Come again?

qkddn49, 2022-08-10:
ellen@hlktech.com

crlogic, 2022-08-10:
👍

qkddn49, 2022-08-10:
May I have your email address?
HI-Link is going to request an espome code for HLK-LD2410.
He's asking me.

crlogic, 2022-08-10:
***@***.***

I am working on ESPHome code and want to do the same for HLK -->
https://community.home-assistant.io/t/mmwave-presence-detection-esphome-style/382778

patrick3399, 2023-05-18:
LD1115 is a leagency module.
Close support.

closed 2023-05-18 by patrick3399