patrick3399/Hi-Link_mmWave_Radar_ESPHome #6: Issue about changing 'unkown' state

JayChun, 2022-09-19:
Hello Patrick,
Thanks for your great component.
I'm using 3 devices which is combined 1115h and 8266 pro.

An issue is that State sensing often turns 'unknown' at one device of them.
Please see below screen-shot.

[image]

- Movement : unknown
- Occupancy : Clearance
- Both : unknown

ESPhome log also same result with dashboard as below.

`INFO Reading configuration /config/esphome/mmwave-1115h-02.yaml...

INFO Starting log output from esp-radar-02.local using esphome API
INFO Successfully connected to esp-radar-02.local
[12:39:53][I][app:102]: ESPHome version 2022.8.3 compiled on Sep 17 2022, 21:37:24
[12:39:53][C][status_led:019]: Status LED:
[12:39:53][C][status_led:020]:   Pin: GPIO2

[12:39:53][C][wifi:491]: WiFi:
[12:39:53][C][wifi:358]:   Hostname: 'esp-radar-02'
[12:39:53][C][wifi:360]:   Signal strength: -85 dB ▂▄▆█
[12:39:53][C][wifi:364]:   Channel: 11

[12:39:53][C][logger:275]: Logger:
[12:39:53][C][logger:276]:   Level: DEBUG
[12:39:53][C][logger:277]:   Log Baud Rate: 0
[12:39:53][C][logger:278]:   Hardware UART: UART0
[12:39:53][C][uart.arduino_esp8266:102]: UART Bus:
[12:39:53][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[12:39:53][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[12:39:53][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[12:39:53][C][uart.arduino_esp8266:108]:   Baud Rate: 115200 baud
[12:39:53][C][uart.arduino_esp8266:109]:   Data Bits: 8
[12:39:53][C][uart.arduino_esp8266:110]:   Parity: NONE
[12:39:53][C][uart.arduino_esp8266:111]:   Stop bits: 1
[12:39:53][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[12:39:53][C][template.number:050]: Template Number 'ESP Radar-02 LD1115H TH1'
[12:39:53][C][template.number:050]:   Icon: 'mdi:cogs'
[12:39:53][C][template.number:051]:   Optimistic: YES
[12:39:53][C][template.number:052]:   Update Interval: 60.0s
[12:39:53][C][template.number:050]: Template Number 'ESP Radar-02 LD1115H TH2'
[12:39:53][C][template.number:050]:   Icon: 'mdi:cogs'
[12:39:53][C][template.number:051]:   Optimistic: YES
[12:39:53][C][template.number:052]:   Update Interval: 60.0s
[12:39:53][C][template.number:050]: Template Number 'ESP Radar-02 LD1115H Clearence Time'
[12:39:53][C][template.number:050]:   Icon: 'mdi:cogs'
[12:39:53][C][template.number:051]:   Optimistic: YES
[12:39:53][C][template.number:052]:   Update Interval: 60.0s
[12:39:53][C][template.number:050]: Template Number 'ESP Radar-02 LD1115H Movement Time'
[12:39:53][C][template.number:050]:   Icon: 'mdi:cogs'
[12:39:53][C][template.number:051]:   Optimistic: YES
[12:39:53][C][template.number:052]:   Update Interval: 60.0s
[12:39:53][C][template.sensor:023]: Template Sensor 'ESP Radar-02 LD1115H Spectral line'
[12:39:53][C][template.sensor:023]:   State Class: ''
[12:39:53][C][template.sensor:023]:   Unit of Measurement: ''
[12:39:53][C][template.sensor:023]:   Accuracy Decimals: 0
[12:39:53][C][template.sensor:023]:   Icon: 'mdi:radar'
[12:39:53][C][template.sensor:024]:   Update Interval: 60.0s
[12:39:53][C][template.sensor:023]: Template Sensor 'ESP Radar-02 LD1115H Signal Strength'
[12:39:53][C][template.sensor:023]:   State Class: ''
[12:39:53][C][template.sensor:023]:   Unit of Measurement: ''
[12:39:53][C][template.sensor:023]:   Accuracy Decimals: 0
[12:39:53][C][template.sensor:023]:   Icon: 'mdi:signal-distance-variant'
[12:39:53][C][template.sensor:024]:   Update Interval: 60.0s
[12:39:53][C][template.text_sensor:021]: Template Sensor 'ESP Radar-02 LD1115H Occupancy Status'
[12:39:53][C][template.text_sensor:021]:   Icon: 'mdi:motion-sensor'
[12:39:53][C][template.binary_sensor:018]: Template Binary Sensor 'ESP Radar-02 LD1115H Occupancy or Movement'
[12:39:53][C][template.binary_sensor:018]:   Device Class: 'occupancy'
[12:39:53][C][template.binary_sensor:018]: Template Binary Sensor 'ESP Radar-02 LD1115H Movement'
[12:39:53][C][template.binary_sensor:018]:   Device Class: 'motion'
[12:39:53][C][mdns:094]: mDNS:
[12:39:53][C][mdns:095]:   Hostname: esp-radar-02
[12:39:53][C][ota:085]: Over-The-Air Updates:
[12:39:53][C][ota:086]:   Address: esp-radar-02.local:8266
[12:39:53][C][api:138]: API Server:
[12:39:53][C][api:139]:   Address: esp-radar-02.local:6053
[12:39:53][C][api:143]:   Using noise encryption: NO
[12:39:53][C][serial.text_sensor:029]: Serial Sensor 'ESP Radar-02 LD1115H UART Text'
[12:39:53][C][serial.text_sensor:029]:   Icon: 'mdi:format-text'
[12:40:04][D][text_sensor:067]: 'ESP Radar-02 LD1115H Occupancy Status': Sending state 'Clearance'
[12:41:04][D][text_sensor:067]: 'ESP Radar-02 LD1115H Occupancy Status': Sending state 'Clearance'
[12:42:04][D][text_sensor:067]: 'ESP Radar-02 LD1115H Occupancy Status': Sending state 'Clearance'
[12:43:04][D][text_sensor:067]: 'ESP Radar-02 LD1115H Occupancy Status': Sending state 'Clearance'
[12:44:04][D][text_sensor:067]: 'ESP Radar-02 LD1115H Occupancy Status': Sending state 'Clearance'
[12:45:04][D][text_sensor:067]: 'ESP Radar-02 LD1115H Occupancy Status': Sending state 'Clearance'
[12:46:04][D][text_sensor:067]: 'ESP Radar-02 LD1115H Occupancy Status': Sending state 'Clearance' `

I try to change some configuration numbers of "movement_time","clearence_time","th_1","th_2" on dashboard, but the device doesn't wake up.
The device wake up when I (several times) enter the space which is placed device.
I wonder what is the problem. Do you have any way to wake up device through HA automation or something?

Inkblot720, 2023-02-10:
Were you able to figure anything out?

patrick3399, 2023-02-10:
It might happend because UART Cable might have some problem.

nickelnine, 2023-02-27:
I'm facing similar issue with MH-ET LIVE ESP32 board.
`Using 'COM8' as serial port.
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting......
Detecting chip type... ES96 %)Writing at 0x000c2946... (100 %)Wrote 743904 bytes (484366 compressed) at 0x00010000 in 11.8 seconds (effective 504.6 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[15:27:00]I (309) cpu_start: Pro cpu up.
[15:27:00]I (309) cpu_start: Starting app cpu, entry point is 0x40081d00
[15:27:00]I (0) cpu_start: App cpu up.
[15:27:00]I (323) cpu_start: Pro cpu start user code
[15:27:00]I (323) cpu_start: cpu freq: 160000000
[15:27:00]I (323) cpu_start: Application information:
[15:27:00]I (328) cpu_start: Project name:     esp-radar-office
[15:27:00]I (333) cpu_start: App version:      2022.11.5
[15:27:00]I (339) cpu_start: Compile time:     Feb 27 2023 15:03:36
[15:27:00]I (345) cpu_start: ELF file SHA256:  e4fdf2427496810e...
[15:27:00]I (351) cpu_start: ESP-IDF:          4.3.2
[15:27:00]I (356) heap_init: Initializing. RAM available for dynamic allocation:
[15:27:00]I (363) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
[15:27:00]I (369) heap_init: At 3FFB8460 len 00027BA0 (158 KiB): DRAM
[15:27:00]I (375) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[15:27:00]I (381) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[15:27:00]I (388) heap_init: At 4009485C len 0000B7A4 (45 KiB): IRAM
[15:27:00]I (395) spi_flash: detected chip: generic
[15:27:00]I (399) spi_flash: flash io: dio
[15:27:00]I (404) cpu_start: Starting scheduler on PRO CPU.
[15:27:00]I (0) cpu_start: Starting scheduler on APP CPU.
[15:36:01]ets Jun  8 2016 00:22:57
[15:36:01]
[15:36:01]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[15:36:01]configsip: 0, SPIWP:0xee
[15:36:01]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[15:36:01]mode:DIO, clock div:2
[15:36:01]load:0x3fff0018,len:4
[15:36:01]load:0x3fff001c,len:1044
[15:36:01]load:0x40078000,len:8896
[15:36:01]load:0x40080400,len:5828
[15:36:01]entry 0x400806ac
`
In Home Assistant, all Sensors are in 'Unknown' state.

patrick3399, 2023-02-28:
> I'm facing similar issue with MH-ET LIVE ESP32 board. `Using 'COM8' as serial port. Connecting...... Detecting chip type... Unsupported detection protocol, switching and trying again... [...]

Check your ESP flash partition, maybe wrong Flash size.

patrick3399, 2023-02-28:
> Hello Patrick, Thanks for your great component. I'm using 3 devices which is combined 1115h and 8266 pro.
>
> An issue is that State sensing often turns 'unknown' at one device of them. Please see [...]

Have to enable UART debug option. Maybe some error on UART receive.

closed 2023-05-18 by patrick3399