patrick3399/Hi-Link_mmWave_Radar_ESPHome #7: Settings (TH1 TH2, Fading times) don't get saved

softhuntermike, 2022-10-22:
Whenever the sensor is restarted (eg.unplugged then replugged), TH1, TH2, fading settings get reset to original (TH1=120, TH2=250, Clearence Time = 5, Movement time =1).

softhuntermike, 2022-10-22:
It doesn't write the new settings it seemed. After adjusting, a refresh will reset everything back

patrick3399, 2022-10-23:
In template number, if restore_value set to true, it will restore the value.

r34220, 2023-01-18:
I see the same. restore_value is set to true but nothing is saved.

Surgikill, 2024-04-17:
I have this same issue. Is there any fix for it?

Surgikill, 2024-04-18:
I fixed this issue. 

https://github.com/Surgikill/LD1125h_esp32/blob/main/presencebox/presencebox_esphome.yaml

There needs to be a delay between each uart.write message or else it floods the uart and the presence sensor cannot receive the messages.