patrick3399/Hi-Link_mmWave_Radar_ESPHome #10: LD2410 config?

mochoandre, 2022-11-10:
Is it possible to have access to LD2410 Config?

patrick3399, 2022-11-11:
In development

Electronlibre2012, 2022-11-17:
Thanks @patrick3399 for your awesome work!

i use a LD1115H and it works like a charm!

LD2410H are a lot cheaper, so i hope you will succeed in the dev part ;)

is it in on a good way or you think it will be complicated to achieve the dev part please?

markmghali, 2023-02-24:
I am also curious about this! I have both ld1115h and ld2410-C. I know ld2410 is now officially part of esphome. Still curious what you come up with patrick.

Thank you again for your amazing work

patrick3399, 2023-02-28:
> I am also curious about this! I have both ld1115h and ld2410-C. I know ld2410 is now officially part of esphome. Still curious what you come up with patrick.
> 
> Thank you again for your amazing work

Official Support is good. I will give some suggestion on official LD2410 branch.

patrick3399, 2023-05-18:
ESPHome Offical support already.
And Home Assistant Direct Bluetooth integration available.

closed 2023-05-18 by patrick3399