patrick3399/Hi-Link_mmWave_Radar_ESPHome #12: With esp32 and cem5825, unable to use tihe cod ”time: “ in yaml file.

book12345678, 2023-02-15:
Thanks for your code. When I make it with esp32 and CEM5825F, because I need to display the time, I add the following code in yaml, but there will be errors in compilation, and I ask you how to solve it.
time:
   - platform: home assistant
     id: homeassistant_time

The error message when compiling is:

/config/esphome/display-cem5825f-mini32-zhuwo.yaml: In function 'void setup()':
/config/esphome/display-cem5825f-mini32-zhuwo.yaml:37:62: error: reference to 'time' is ambiguous
    board: wemos_d1_mini32 #esp32dev
                                                               ^
In file included from src/esphome/components/homeassistant/time/homeassistant_time.h:4,
                  from src/esphome.h:23,
                  from src/main.cpp:3:
src/esphome/components/time/real_time_clock.h:11:11: note: candidates are: 'namespace esphome::time { }'
  namespace time {
            ^~~~
In file included from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/time.h:22,
                  from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/sys/time.h:414,
                  from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/sys/time.h:16,
                  from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/pthread.h:18,
                  from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/xtensa-esp32-elf/no-rtti/bits/gthr-default.h:48,
                  from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/xtensa-esp32-elf/no-rtti/bits/gthr.h:151,
                  from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/ext/atomicity.h:35,
                  from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:39,
                  from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/string:52,
                  from src/esphome/components/socket/socket.h:2,
                  from src/esphome/components/api/api_frame_helper.h:13,
                  from src/esphome/components/api/api_connection.h:3,
                  from src/esphome.h:3,
                  from src/main.cpp:3:
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/time.h:59:11: note: 'time_t time(time_t*)'
  time_t time (time_t *_timer);
            ^~~~
/config/esphome/display-cem5825f-mini32-zhuwo.yaml:45:66: error: reference to 'time' is ambiguous
        type: git
                                                                   ^
In file included from src/esphome/components/homeassistant/time/homeassistant_time.h:4,
                  from src/esphome.h:23,
                  from src/main.cpp:3:
src/esphome/components/time/real_time_clock.h:11:11: note: candidates are: 'namespace esphome::time { }'
  namespace time {
            ^~~~
In file included from /data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/time.h:22,
                  from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/sys-include/sys/time.h:414,

patrick3399, 2023-02-15:
The time in the code is not compatible with time components.
Need to rewrite the timing code to solve this problem.

book12345678, 2023-02-15:
Thank you for your reply, the code is already very good. It would be even better if it could be compatible with the time components. Looking forward to the adjusted code.

patrick3399, 2023-02-24:
I will rewrite the code. But i can't promise if it will be done in a few days.

petr-simek, 2023-07-30:
I changed 
```
type: time_t ->  type: unsigned long   (globals components)
time(NULL) -> millis()   (text_sensor LD1115H_UART_BUS)
```
and it works correctly with the time component

tomaae, 2023-09-14:
Thanks for that, here is my modification. I changed logic to use just binary sensors for ease of use in HA and added some features for displaying entities in proper categories, etc.

```
substitutions:
  devicename: esp_dev
  human_devicename: ESP Dev

packages:
  <<: !include_dir_named common

esphome:
  name: $devicename
  platform: ESP8266
  board: d1_mini
  on_boot:  #LD1115H Initial Setting
    priority: -200
    then:
      - uart.write:
          id: LD1115H_UART_BUS
          data: !lambda |-
            std::string th1st = "th1=" + str_sprintf("%.0f",id(LD1115H_TH1).state) +" \n";
            return std::vector<uint8_t>(th1st.begin(), th1st.end());
      - uart.write:
          id: LD1115H_UART_BUS
          data: !lambda |-
            std::string th2st = "th2=" + str_sprintf("%.0f",id(LD1115H_TH2).state) +" \n";
            return std::vector<uint8_t>(th2st.begin(), th2st.end());

external_components:
  - source:
      type: git
      url: https://github.com/ssieb/custom_components #Thanks for @ssieb components.
    components: [ serial ]

logger:
  level: info  #You Can Use "INFO" Level
  baud_rate: 0

uart:
  id: LD1115H_UART_BUS
  rx_pin: GPIO3  #ESP8266 UART_0
  tx_pin: GPIO1  #ESP8266 UART_0
  baud_rate: 115200
  data_bits: 8
  stop_bits: 1
  parity: NONE

globals:
  - id: LD1115H_Last_Mov_Time
    type: unsigned long
    restore_value: no
    initial_value: "0"
  - id: LD1115H_Last_Occ_Time
    type: unsigned long
    restore_value: no
    initial_value: "0"

interval:
  - interval: 1s #Clearance Scan Time
    setup_priority: -200
    then:
      lambda: |-
        if ((millis()-id(LD1115H_Last_Occ_Time))>(id(LD1115H_Clear_Time).state * 100)) {
          if (id(LD1115H_Occ_Binary).state == true) {
            id(LD1115H_Occ_Binary).publish_state(false);
          }
          if (id(LD1115H_Occ_Binary).state == false && id(LD1115H_Mov_Binary).state == false) {
            id(LD1115H_MovOcc_Binary).publish_state(false);
          }
        }
[... 112 lines cut ...]
          id(LD1115H_Direction).publish_state(atof(id(LD1115H_UART_Text).state.substr(5,2).c_str()));
          if (id(LD1115H_Mov_Binary).state == false) {
            id(LD1115H_Mov_Binary).publish_state(true);
          }
          if (id(LD1115H_MovOcc_Binary).state == false) {
            id(LD1115H_MovOcc_Binary).publish_state(true);
          }
          id(LD1115H_Last_Mov_Time) = millis();
        }

binary_sensor:
  - platform: template
    name: ${human_devicename} Occupancy or Movement Status
    id: LD1115H_MovOcc_Binary
    device_class: presence
  - platform: template
    name: ${human_devicename} Occupancy Status
    id: LD1115H_Occ_Binary
    icon: "mdi:motion-sensor"
    device_class: occupancy
  - platform: template
    name: ${human_devicename} Movement Status
    id: LD1115H_Mov_Binary
    device_class: motion
```