patrick3399/Hi-Link_mmWave_Radar_ESPHome #13: ESPHOME 2023.3.0  esp32 LD1115h Error

qkddn49, 2023-03-16:
INFO Reading configuration /config/esphome/ld1115h.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing ld1115h (board: nodemcu-32s; framework: espidf; platform: platformio/espressif32 @ 5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50001.0 (5.0.1) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /config/esphome/.esphome/build/ld1115h/sdkconfig.ld1115h
-- Configuring incomplete, errors occurred!
See also "/data/ld1115h/.pioenvs/ld1115h/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/tool_version_check.cmake:36 (message):
  

  Tool doesn't match supported version from list ['esp-2022r1-11.2.0']:
  /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc

  Please try to run 'idf.py fullclean' to solve it.

Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/components/esp_common/project_include.cmake:10 (check_expected_tool_version)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:380 (include)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:599 (__build_process_project_includes)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 5.79 seconds ==========================

patrick3399, 2023-03-16:
Just commet ESP-IDF version
like this>
#version: latest

qkddn49, 2023-03-16:
ok.  Thank you.

closed 2023-05-18 by patrick3399