patrick3399/Hi-Link_mmWave_Radar_ESPHome #14: LD2420 Support

soubhik-khan, 2023-04-13:
Requesting support for [LD2420](https://www.aliexpress.us/item/3256805079796034.html?spm=a2g0o.productlist.main.1.3ca41394srJi9Y&algo_pvid=88482b74-7a5f-4fa0-8f6f-1a608e29bcd9&algo_exp_id=88482b74-7a5f-4fa0-8f6f-1a608e29bcd9-0&pdp_npi=3%40dis%21USD%213.83%213.26%21%21%21%21%21%40211be54b16811001858582618d074b%2112000032416762459%21sea%21US%213325717729&curPageLogUid=ojftg9qls3SG)

patrick3399, 2023-05-18:
Will try tofigure out how it works.

vuquanhp, 2023-07-24:
I have bought 2 pcs. And both of them did not run. Use original LD2420 tool on PC, it cannot connect. OT1, OT2 and RX always in high state. Electric consumable is ~50mA - normal, as the datasheet.

simonyel, 2023-08-24:
> I have bought 2 pcs. And both of them did not run. Use original LD2420 tool on PC, it cannot connect. OT1, OT2 and RX always in high state. Electric consumable is ~50mA - normal, as the datasheet.

I got 5 on AliExpress and the same issue, mfg site has no tool or manual posted which is odd. I wonder if these even shipped with firmware loaded.

simonyel, 2023-10-13:
> Some observations: https://twitter.com/andresvahter/status/1712460762206372032

This note did the trick for me! Thank you!

andresv, 2023-10-13:
At least it responds to version query, but I haven't got reasonable sensor output yet.

I am using these resources:
* https://github.com/soubhik-khan/HLK-LD2420
* https://github.com/xuhuan/ld2420/blob/main/ld2420.cpp

Just started to try out scary chinesium tool and logging with the saleae at the same time.

janwego, 2023-10-14:
> Some observations: https://twitter.com/andresvahter/status/1712460762206372032

Thanks Andres, really helped me in getting it to work.
When connecting the device via USB Serial adapter and using the tool, I can see the live image from the device and it changes when I wave the hand in front of it. But when I only connect vial Serial it always says "Range 11 ON" and the Output pin (the correct one) is always high although nothing moved in front of the sensor.

Does anybody know what the values in the Tool mean?

BR,

Jan

janwego, 2023-10-15:
Okay, I solved it:

1. You need to keep in mind what @andresv wrote, pin 3 and 5 are mixed and Baud is 115200
2. I luckily ordered the UART from HiLink ( HLK-CH340E ) so I could connect a LD2420 and read its config with the HLK tool.
I made a screenshot and translated it with google translate so I could understand the parameter at least a little bit.
The problem why others only receive HIGH is: the delay-time (after the movement stopped) is 30 (THIRTY) seconds, every little movement adds again 30 seconds:

[image]

3. To resolve this, I changed the value to 5 seconds and wrote the config to my device. All other sensitivity values were left untouched.
4. The problem I described above (although movement stopped, device keeps sending "Range 11 ON" or "Range 12 ON" happens when you write config to the device. Don't know why but when I pull the usb-adapter out and insert it back again the device acts just as it should with the configured values. 
5. So reboot sensor after config write.
6. Afterwards I could read the distance in cm from the console and after I stopped moving, five seconds later the output switched to "Off" until I moved again

andresv, 2023-10-16:
@janwego how did you read out distance?
I put sensor into streaming mode using hlk tool and I can see that following data is sent periodically (around 128 bytes). However it seems to have totally different header.

[image]

janwego, 2023-10-16:
You were faster than me - glad it works now for you :)
Don't even know what you meant by streaming mode - I just connected the device to the Serial Adapter and connected with Putty to COM-Port and Baud 115200

andresv, 2023-10-16:
If you start this:
[image]
Then sensor starts to send ~126 bytes data packets to UART. Here it is mentioned that this mode is not documented anywhere: https://github.com/descipher/esphome/blob/24fa4147dd50f612dd284ef4647f42def0deab10/esphome/components/ld2420/ld2420.cpp#L134 so currently it seems that just HLK tool knows about it.

janwego, 2023-10-16:
Aaah okay, now I know what you mean - I meant that when I said "I can see the live image from the device" . Don't know if they'll ever deliver the code. The LD2420 isn't completely supported by ESPHome yet is it?

jcard0na, 2023-12-07:
Hi @andresv, @janwego 

Were you able to decipher the LD2420 communication?  I'm trying to write a driver for it, but the data I get from the device does not match the protocol documented [here](https://github.com/soubhik-khan/HLK-LD2420/blob/main/HLK-LD2420%20command%20protocol.xlsx)

Instead, I'm getting ASCII output like the one you observed:

```
Sent: [fd, fc, fb, fa, 4, 0, ff, 0, 2, 0, 4, 3, 2, 1]
Wait 100ms
Clear input buffer
Sent: [fd, fc, fb, fa, 4, 0, ff, 0, 2, 0, 4, 3, 2, 1]
Received: [4f, 4e, d, a, 52, 61, 6e, 67, 65, 20, 32, 35, d, a]
(in ASCII): 
ON
Range 25
```

The protocol documentation says I should expect a binary reply like `[FD FC FB FA 08 00 FF 01 00 00 02 00 20 00 04 03 02 01]`

I'll appreciate any pointers you can share!  I've reached out to the manufacturer who have not been very responsive so far...

andresv, 2023-12-07:
I did not look into deciphering and just went with ASCII output.

eried, 2024-01-01:
> Hi @andresv, @janwego
>
> Were you able to decipher the LD2420 communication? I'm trying to write a driver for it, but the data I get from the device does not match the protocol documented [...]

maybe the protocol is the same as ld2410 instead?

@andresv I wonder if is the ld2420 the same with regards as reporting gates/energy in "engineering mode" as the ld2410? (as the later reports 9 values of static + 9 values of movement, I wonder if the ld2420 reports 15 of each)

patrick3399, 2024-01-01:
Here is the reference.
https://github.com/esphome/esphome/tree/dev/esphome/components/ld2420

simonyel, 2024-01-01:
There's an official component support for LD2420

Sent from mobile. Please pardon any typos

perlchamp, 2024-01-02:
Hi, what we're talking about?
What Version of board you have ?
Mine is V2.1, the tool you need for that is V1.2.1.1 AND manual V1.3 from december '23.
I've 'modified' this tool, so it's now in 'denglisch' language (german-english-mix).
[GER__HLK-LD2420_TOOL.zip](https://github.com/patrick3399/Hi-Link_mmWave_Radar_ESPHome/files/13807125/GER__HLK-LD2420_TOOL.zip)
But i can't find the manual for it. It has to be V1.3 manual, please in englisch or german.
The chinese version of the 1.3 Manual has inserted some jpeg instead of text so no copying/translating is possible...
Maybe somebody can find a english-version of this manual, thanks.

so long
Pc

jcard0na, 2024-01-02:
Thanks @perlchamp 

I also have board v2.1.  The manual I have is v1.2, and the manufacturer also provided a protocol document without a version number (attached).  

[HLK-LD2420 Protocol Document.pdf](https://github.com/patrick3399/Hi-Link_mmWave_Radar_ESPHome/files/13813436/HLK-LD2420.Protocol.Document.pdf)

perlchamp, 2024-01-03:
Hi, thank you. I have this document, too. 
In a few days i will post the final version of the .exe.

elas123, 2024-01-21:
anyone know the latest firmware? i got  with 1.5.9. is there newer?

ihayri, 2024-04-06:
Does anybody have a library that works with the LD2420, need to get it running on an ESP32 without ESPHome, thanks.

Higgmeister, 2024-05-30:
For those coming here looking for an answer to why they are only seeing text (**ON/OFF Range ##**) when reading the LD2420.

It's a matter of being in the right mode and "Normal Working" mode is the default which is why you see the limited text. Following is a list of modes supported by the LD2420.  The value (I do mine in HEX) must be sent to the LD2420 before reading the data stream.  Once the sensor is put into "Report" mode, you can read the stream and it'll be 45 hex characters long.  As mentioned above, once the sensor is power cycled or reset, it reverts into "Normal Working" mode again and needs to be set again.

This code puts LD2420 into report mode. Must do before reading buffer.
FD FC FB FA 08 00 12 00 00 00 04 00 00 00 04 03 02 01 

This code puts LD2420 into Normal working mode. This outputs detection results and range gate of detected target.
FD FC FB FA 08 00 12 00 00 00 64 00 00 00 04 03 02 01 

This code puts LD2420 into debugging mode.
FD FC FB FA 08 00 12 00 00 00 00 00 00 00 04 03 02 01

jschoch, 2025-03-09:
thanks for this info, i found an engish version of the tool here 

https://h.hlktech.com/Mobile/download/fdetail/225.html

perlchamp, 2025-03-09:
you only have to send an email to hlktech with the wish to get the newest version. After a couple of weeks you will get it via download-link...

straltest, 2025-04-24:
> maybe the protocol is the same as ld2410 instead?
>
> [@andresv](https://github.com/andresv) I wonder if is the ld2420 the same with regards as reporting gates/energy in "engineering mode" as the [...]

Any one able to decipher why the recieved buffer is like [4f, 4e, d, a, 52, 61, 6e, 67, 65, 20, 32, 35, d, a] instead of something like [FD FC FB FA 08 00 FF 01 00 00 02 00 20 00 04 03 02 01]`?

Also, did anybody create the library that works with ESP32 without the ESPHome?

perlchamp, 2025-04-24:
Maybe this could help

[HLK-LD2420---Kommunikationsprotokoll.pdf](https://github.com/user-attachments/files/19897762/HLK-LD2420---Kommunikationsprotokoll.pdf)

Good luck !

etejesh, 2025-04-25:
The sensor is not behaving as documented in the original protocol document which is almost similar to what you have shared. 

So I have asked for the help of those who have worked on this issue, which was also reported earlier here.