patrick3399/Hi-Link_mmWave_Radar_ESPHome #16: Error at compiling time

jcastro, 2023-06-12:
Hi!

I'm using ESPHome 2023.5.5 Add-On on HA running inside a VM. At the time of trying to compile, I'm getting these errors. Trying [with this config ](https://github.com/patrick3399/Hi-Link_mmWave_Radar_ESPHome/blob/main/LD1125H/ESP32-LD1125H-Complete.yaml)

Thanks!

[image]

jcastro, 2023-06-12:
If I use arduino it works ok, but would love to use idf as recommended!

homonto, 2023-06-13:
Same here:
```
INFO ESPHome 2023.5.5
INFO Reading configuration /mnt/disk/scripts/003-esphome/ESP32-LD1125H-Complete.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.0.9
INFO Compiling app...
Processing esp-radar (board: nodemcu-32s; framework: espidf; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50001.0 (5.0.1) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Building ESP-IDF components for target esp32
Processing 1 dependencies:
[1/1] idf (5.0.1)
-- Project sdkconfig file /mnt/disk/scripts/003-esphome/.esphome/build/esp-radar/sdkconfig.esp-radar
-- Configuring incomplete, errors occurred!
See also "/mnt/disk/scripts/003-esphome/.esphome/build/esp-radar/.pioenvs/esp-radar/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at /home/pi/.platformio/packages/framework-espidf/tools/cmake/tool_version_check.cmake:36 (message):
  

  Tool doesn't match supported version from list ['esp-2022r1-11.2.0']:
  /home/pi/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc

  Please try to run 'idf.py fullclean' to solve it.

Call Stack (most recent call first):
  /home/pi/.platformio/packages/framework-espidf/components/esp_common/project_include.cmake:10 (check_expected_tool_version)
  /home/pi/.platformio/packages/framework-espidf/tools/cmake/build.cmake:380 (include)
  /home/pi/.platformio/packages/framework-espidf/tools/cmake/build.cmake:599 (__build_process_project_includes)
  /home/pi/.platformio/packages/framework-espidf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 1.86 seconds ==========================
```

patrick3399, 2023-06-13:
Try to clean build files, maybe can fix the problem.

jcastro, 2023-06-13:
> Try to clean build files, maybe can fix the problem. It is likely a ESP-IDF tool chain error.

I tried it but no luck, I even tried a different Mac!

patrick3399, 2023-06-13:
> > Try to clean build files, maybe can fix the problem. It is likely a ESP-IDF tool chain error.
> 
> I tried it but no luck, I even tried a different Mac that was fresh and no esptool or anything installed.
> 
> I can compile other idf projects like the JK-BMS tho

Maybe try to use ESP-IDF recommand version?
Not using the latest.

digiblur, 2023-06-13:
I'm using 2023.5.5 and just built a fresh one today with this code: https://digiblur.com/2023/05/24/esphome-mmwave-presence-how-to-guide#esphome-code

homonto, 2023-06-13:
> Maybe try to use ESP-IDF recommand version? Not using the latest.

I removed: "version" and it compiles like a charm - thank you

homonto, 2023-06-13:
> I'm using 2023.5.5 and just built a fresh one today with this code: https://digiblur.com/2023/05/24/esphome-mmwave-presence-how-to-guide#esphome-code

thx bro, yours is 1125 - I am trying 1115
plus you have gazyllion other thingis there - sda... ;-)

thx

jcastro, 2023-06-13:
yeah same here! thank you all!

homonto, 2023-06-13:
however going esp-idf instead of arduino brings some unexpected issues: no captive portal, no ota etc

digiblur, 2023-06-13:
> > I'm using 2023.5.5 and just built a fresh one today with this code: https://digiblur.com/2023/05/24/esphome-mmwave-presence-how-to-guide#esphome-code
> 
> thx bro, yours is 1125 - I am trying 1115 plus you have gazyllion other thingis there - sda... ;-)
> 
> thx

Should all be the same for sure... and nothing wrong with extra stuff, shows that compiles too ;)

digiblur, 2023-06-13:
> > Should all be the same for sure... and nothing wrong with extra stuff, shows that compiles too ;)
> 
> ha, today I received 2x 1115 and 2x 1125 I am wondering what is the difference - for sure price there is ;-)

More configuration options on the 1125. I tried a few myself.

patrick3399, 2023-06-13:
> however going esp-idf instead of arduino brings some unexpected issues: no captive portal, no ota etc

ESP-IDF have OTA.

closed 2023-06-13 by jcastro