patrick3399/Hi-Link_mmWave_Radar_ESPHome #17: LD1125H Rmax not being restored

bluegizmo83, 2023-06-25:
The ESPHome code for restore_value for Rmax is set to TRUE, but after every power cycle of the ESP32 it does not restore the value. For example, if I have Rmax set to 2 meters, In Home Assistant the value looks restored after a power cycle as it still shows 2, but the sensor reads beyond 2 meters until I move the slider then set it back to 2 again. I'm not sure if the sensitivity sliders are really restoring their values or not, because adjusting the sensitivity barely seems to do anything anyway, so it's hard to tell for those.

jcastro, 2023-07-16:
Yes! I was about to report this as well. Here's a video showing this behavior. In my case, I don't think I've restarted the ESP32, but it seems to "randomly" start to read more distance than the one that's set. Hope this helps @patrick3399 

https://github.com/patrick3399/Hi-Link_mmWave_Radar_ESPHome/assets/190036/48ebac98-d494-422f-9bf7-57126828493c

patrick3399, 2023-07-17:
II will change the code. And need both of you to test the new code.

jcastro, 2023-07-17:
> II will change the code. And need both of you to test the new code.

for sure! would love to, thanks a lot

patrick3399, 2023-07-22:
https://pastebin.com/meR6aMcs
@jcastro @bluegizmo83 
Here is a test version. Please give me feedback if you use it.

jcastro, 2023-07-23:
> https://pastebin.com/meR6aMcs
> 
> @jcastro @bluegizmo83 
> 
> Here is a test version. Please give me feedback if you use it.

Sure! Do you mind sharing what you changed so I can attach it or change it on my config?

jcastro, 2023-07-23:
@patrick3399 I've tested it but I can't compile, even if I change version to recommended

`esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf  #Suggest Use ESP-IDF Framework, or Plug Out the UART Cable Might Cause ESP32 Hang. 
    version: recommended`

`HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40404.0 (4.4.4) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
No dependencies
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/src/main.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/console/argtable3/arg_file.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/console/argtable3/arg_hashtable.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/console/argtable3/arg_int.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/console/argtable3/arg_lit.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/console/argtable3/arg_rem.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/console/argtable3/arg_rex.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/console/argtable3/arg_str.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/console/argtable3/arg_utils.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/console/argtable3/argtable3.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/cxx/cxx_exception_stubs.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/cxx/cxx_guards.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/gpio.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/i2c.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/i2s.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/ledc.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/pcnt.o
Archiving /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/esp-idf/console/libconsole.a
Archiving /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/esp-idf/cxx/libcxx.a
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/periph_ctrl.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/rmt.o
/config/esphome/salon-occupancy-test.yaml: In lambda function:
/config/esphome/salon-occupancy-test.yaml:93:34: error: invalid operands of types 'const char [5]' and 'const char [3]' to binary 'operator+'
               std::string th2st = "save" + "\r\n";
                           ~~~~~~~^~~~~~~~
/config/esphome/salon-occupancy-test.yaml: In lambda function:
/config/esphome/salon-occupancy-test.yaml:111:19: error: redeclaration of 'std::__cxx11::string th2st'
               std::string th2st = "save" + "\r\n";
                   ^~~~~
/config/esphome/salon-occupancy-test.yaml:110:19: note: 'std::__cxx11::string th2st' previously declared here
               std::string th2st = "mth2=" + str_sprintf("%.0f",x) +"\r\n";
                   ^~~~~
/config/esphome/salon-occupancy-test.yaml:111:34: error: invalid operands of types 'const char [5]' and 'const char [3]' to binary 'operator+'
               std::string th2st = "save" + "\r\n";
                           ~~~~~~~^~~~~~~~
/config/esphome/salon-occupancy-test.yaml: In lambda function:
/config/esphome/salon-occupancy-test.yaml:129:34: error: invalid operands of types 'const char [5]' and 'const char [3]' to binary 'operator+'
               std::string th2st = "save" + "\r\n";
                           ~~~~~~~^~~~~~~~
/config/esphome/salon-occupancy-test.yaml: In lambda function:
/config/esphome/salon-occupancy-test.yaml:147:34: error: invalid operands of types 'const char [5]' and 'const char [3]' to binary 'operator+'
               std::string th2st = "save" + "\r\n";
                           ~~~~~~~^~~~~~~~
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/rtc_io.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/rtc_module.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/sdspi_crc.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/sdspi_host.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/sdspi_transaction.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/sigmadelta.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/spi_common.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/spi_master.o
Compiling /data/salon-occupancy-test/.pioenvs/salon-occupancy-test/driver/spi_slave.o
*** [/data/salon-occupancy-test/.pioenvs/salon-occupancy-test/src/main.o] Error 1
========================== [FAILED] Took 6.53 seconds ==========================`

jcastro, 2023-07-23:
If I use the file as it is, using `latest` for version section in esp32

`INFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/salon-occupancy-test.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.0.9
INFO Compiling app...
Processing salon-occupancy-test (board: nodemcu-32s; framework: espidf; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50001.0 (5.0.1) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Manifest files have changed, solving dependencies.
.Updating lock file at /config/esphome/.esphome/build/salon-occupancy-test/dependencies.lock
Processing 1 dependencies:
[1/1] idf (5.0.1)
-- Project sdkconfig file /config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:193 CONFIG_MCPWM_ISR_IN_IRAM was replaced with CONFIG_MCPWM_ISR_IRAM_SAFE
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:272 CONFIG_ESP32_DEFAULT_CPU_FREQ_80 was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_80
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:273 CONFIG_ESP32_DEFAULT_CPU_FREQ_160 was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_160
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:274 CONFIG_ESP32_DEFAULT_CPU_FREQ_240 was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:275 CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ was replaced with CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:276 CONFIG_ESP32_SPIRAM_SUPPORT was replaced with CONFIG_SPIRAM
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:279 CONFIG_ESP32_ULP_COPROC_ENABLED was replaced with CONFIG_ULP_COPROC_ENABLED
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:280 CONFIG_ESP32_ULP_COPROC_RESERVE_MEM was replaced with CONFIG_ULP_COPROC_RESERVE_MEM
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:281 CONFIG_ESP32_DEBUG_OCDAWARE was replaced with CONFIG_ESP_DEBUG_OCDAWARE
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:282 CONFIG_ESP32_BROWNOUT_DET was replaced with CONFIG_ESP_BROWNOUT_DET
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:283 CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_0 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_0
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:284 CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_1 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_1
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:285 CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_2 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_2
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:286 CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_3 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_3
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:287 CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_4 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_4
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:288 CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_5 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_5
/config/esphome/.esphome/build/salon-occupancy-test/sdkconfig.salon-occupancy-test:289 CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_6 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_6
[... cut ...]

Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/components/esp_common/project_include.cmake:10 (check_expected_tool_version)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:380 (include)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:599 (__build_process_project_includes)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:3 (project)`

patrick3399, 2023-07-23:
@jcastro Here is a new version. It can be compile at ESPHome 2023.7.0
https://pastebin.com/rBGWHxPn

jcastro, 2023-07-23:
> @jcastro Here is a new version. It can be compile at ESPHome 2023.7.0 https://pastebin.com/rBGWHxPn

Thanks! this one compiled correctly. Testing the new code as we speak!

bluegizmo83, 2023-07-23:
> @jcastro Here is a new version. It can be compile at ESPHome 2023.7.0 https://pastebin.com/rBGWHxPn

I just tested this version as well. Still has the same problem. I set the RMax to 1 meter for testing, and it stops detecting at 1 meter as it should, but then as soon as the sensor is power cycles, it starts reading well passed 1 meter again, like its not remembering the setting, even though Home Assistant still shows the RMax value is set a 1 meter.

jcastro, 2023-07-23:
> 

I've tried to restart it (using the ESPHome restart button) and it seems to save and restore it correctly. How are you powering off and on the ESP32 board? I will try to reproduce it too!

bluegizmo83, 2023-07-23:
> > 
> 
> I've tried to restart it (using the ESPHome restart button) and it seems to save and restore it correctly. How are you powering off and on the ESP32 board? I will try to reproduce it too!

I'm currently powering the ESP32 board via the usb port with a power bank. Unplugging the usb power and plugging it back in is how I've been power cycling it.

patrick3399, 2023-07-23:
Oh，There is a ESPHome setting to prevent the Flash overwrite, so make the change and wait 3 minute to restart. Maybe it can make the setting correctly.

Surgikill, 2024-04-17:
I seem to be having the same issue. Is there any official fix/update for this?

kotylo, 2025-02-23:
If anyone struggling, here is the fix (add `-delay: 100ms` in on_boot section before every `uart` command):
https://github.com/patrick3399/Hi-Link_mmWave_Radar_ESPHome/issues/7#issuecomment-2062875850