patrick3399/Hi-Link_mmWave_Radar_ESPHome #18: Two LD1125H in one ESP32?

jcastro, 2023-06-29:
Hey guys!

Have you tried to run 2 LD1125H sensors into one ESP32 module? Just curious to hear about your experience. Thanks!

patrick3399, 2023-06-30:
You can test it yourself.
Suggestion use the hardware UART.

jcastro, 2023-07-01:
For sure! Probably will test it today and report back

jcastro, 2023-07-01:
IT WORKS! I will leave it a few days to see if it's stable and report back

jcastro, 2023-07-04:
Yup! Really stable so far, but looks like putting 2 sensors side by side have some interferences

patrick3399, 2023-07-05:
Nice. Because I only have one LD1125H, so hard to test duo radar at same ESP device.

jcastro, 2023-07-05:
> 

Would love to donate some sensors if it’s helpful for you to test or add support (I heard there’s a new LD2450!)

patrick3399, 2023-07-05:
> > 
> 
> Would love to donate some sensors if it’s helpful for you to test or add support (I heard there’s a new LD2450!)

Actually I already have LD2450 and LD2420 model, but there is no hurry to develope the code because we have few model to properly work.