patrick3399/Hi-Link_mmWave_Radar_ESPHome #19: using HD1125 on Atom Lite (software serial in esp-idf, on atom lite?)

netweaver1970, 2023-06-30:
Hi,
Sorry for abusing the 'issue' form, but I didn't have another public way for interacting.

I would like to interface my HD1125 to my Atom Lite ESP32-Pico-D4 equipped MCU. [https://docs.m5stack.com/en/core/atom_lite](https://docs.m5stack.com/en/core/atom_lite)

But none of the additional uart pins (for port uart1/2) are broken out, so I'll need to go the software serial route.
In esphome, the arduino framework, we can use the software serial and use any pin available but for the esp-idf I'm not sure if that's possible. Any suggestion/idea?
I saw this [https://github-com.translate.goog/junhuanchen/esp-idf-software-serial?_x_tr_sl=auto&_x_tr_tl=zh-CN&_x_tr_hl=en-US&_x_tr_pto=wapp](https://github-com.translate.goog/junhuanchen/esp-idf-software-serial?_x_tr_sl=auto&_x_tr_tl=zh-CN&_x_tr_hl=en-US&_x_tr_pto=wapp), a software serial lib for esp-idf, but it doesn't look easily integratable into esphome (my uneducated guess, pls correct if wrong) and before going into that rabbit hole, I rather check here 1st, maybe someone faced this issue before, as most of the sample yaml code is esp-idf based.
Thanks!

patrick3399, 2023-06-30:
Actually. ESP32 hardward UART is not specify to any pin.
The internal hardware UART receiver can be set at any pins.

patrick3399, 2023-06-30:
UART 0,1,2 pins are default. But you can change it.

netweaver1970, 2023-07-01:
mea culpa, it seems to 'just work' as well with the Atom Lite. When putting my scope on the TX pin I see negative going (5V->0V) pulses appearing when changing the mthx and rmax device sliders in HA. So all should be fine (fingers crossed) after proper connecting the mmWave module. Closing issue. 

_Duh, I forgot to click the 'comment' button, I wrote this closing comment 30 min after opening the issue :( - 
Anyway, extra update: all works fine indeed! Thanks a lot for the hard work in coding/decoding the messages and creating all esphome yamls for these nice nifty sensors_