patrick3399/Hi-Link_mmWave_Radar_ESPHome #25: Board type framework

withanhdammit, 2023-10-12:
In your sample YAML, you suggest using the esp-idf framework
`  framework:
    type: esp-idf  #Suggest Use ESP-IDF Framework, or Plug Out the UART Cable Might Cause ESP32 Hang. `

What exactly are the ramifications of using the 'arduino' framework?  I'm attaching a BME680 and I want to use the 'bme680_besc', if I use the esp-idf framework, it throws an error saying the BME680 requires the Arduino framework.  I switched mine over to 'arduino', but it seems like the motion/occupancy is still working.

patrick3399, 2023-10-16:
There is a bug in the Arduino Framework before.
So If you test the code and need to used the arduino framework, just change it.

withanhdammit, 2023-10-19:
Thank you!  Hopefully the arduino framework bug doesn't affect me.

closed 2023-10-19 by withanhdammit