patrick3399/Hi-Link_mmWave_Radar_ESPHome #26: Is there any way to decrease the sesitivity of thie device?

bert269, 2023-10-16:
I ahve tried to rut mth2 & mth3 all the way down - but it does not seem to make any difference.
Also, I want to used this inside an pantry to swith the light on and off if there are movement inside the panty - but it picks up movement from far away, though the walls.
I place a foild cover on the sides and back of this sensor -to limit the signals to be so wide and more directional, but it does not seem to work at all.
I ahve the setting at:
Clearance time: 5.0
Movement Time: 1.0
mth1: 60
mth2:30
mth3:20
rmax: 2.0

patrick3399, 2023-10-16:
Put a plastic sheet on the front of the device can reduce the sensitive.

bert269, 2023-10-16:
I have it behind a wallplate outlet cover. I do not know if this is
sufficient plastic?

Can you please tell me, if I do not want mth2 and mth3 to be active, do I
slide it to left all the way - or to the right?. Since this is inside a
pantry (small area) I think mth1 should be sufficient? I have that set at 1.
Also, what should I set mth-max as? I have it currently at 1.

Thanks for your time and help.